muccc/iridium-toolkit #28: Satellite name doesn't match with Celestrak TLE data

a-dz, 2020-05-27:
Hi! 
ra_sat field of RA message doesn't correspond to TLE file of Celestrak, and satellite monitoring (http://celestrak.com/NORAD/elements/). How should I handle that?

schneider42, 2020-05-27:
The SV number in the Ring Alert messages is a 7 bit field. Originally it simply contained the `IRIDIUM XY` number of Block 1 SVs.

With the introduction of Iridium Next, this field became to small (`66 * 2 < 2**7`), so Iridium started to reuse numbers of (de-activated / de-orbited) Block 1 SVs. This means that the SV number in the Ring Alert message does not equal the `IRIDIUM XYZ` number anymore.

We haven't looked into this anymore, but it should be trivial to create a table which provides a mapping from SV number in the Ring Alert to "IRIDIUM XYZ" designation. Wanna try? ;)

Sec42, 2022-02-03:
I just pushed code to do this in 2e3d1aa5bcd6b8fc9bc51fae620442c600d73516.

You will need to install the python3 skyfield module and have TLE data in tracking/iridium-NEXT.txt - which you can fetch/update that by running `cd tracking && ./update-tracking`

Pipe some parsed IRA into `reassembler.py -m satmap` and it should output a mapping for those based on TLE data.

Please note that the TLE data and your received IRA frames need to be the same age. If they differ by more then a couple of days, accuracy of the mapping may be reduced.

closed 2022-02-27 by Sec42