muccc/iridium-toolkit #31: reassembler page mode error

avdsdr, 2020-08-05:
When reassembler.py is called with the mode specified as page, we receive the following error.  We are able to extract the contents for the other modes and voice.  This was run on Ubuntu 18.04 with the most recent code in master.

$ ./reassembler.py -i ../output.parsed -m page
Traceback (most recent call last):
  File "./reassembler.py", line 458, in <module>
    zx.run(fileinput.input(ifile))
  File "./reassembler.py", line 86, in run
    zz=self.process(res)
  File "./reassembler.py", line 344, in process
    return ["%02d %02d %s %s %03d : %s %s"%(q.sat,q.beam,q.posx,q.posy,q.alt,x[0],x[1])]
AttributeError: 'MyObject' object has no attribute 'posx'

Sec42, 2020-08-05:
Thanks for the report. Fixed in e9ff31e.

closed 2020-08-05 by Sec42

avdsdr, 2020-08-05:
Verified the fix.

Thank you Sec42!