muccc/iridium-toolkit #33: iridium-parser : TypeError: argument 1 must be string or read-only buffer, not list

Geezerjem, 2020-08-31:
On a new install, I'm getting :

jem@jem-VBMINT18 ~/Dev/iridium-toolkit $ python iridium-parser.py ../../iridium/output-err.bits > ../../iridium/output-err.parsed
Traceback (most recent call last):
  File "iridium-parser.py", line 1602, in <module>
    do_input(input)
  File "iridium-parser.py", line 1516, in do_input
    perline(Message(line.strip()).upgrade())
  File "iridium-parser.py", line 268, in upgrade
    return IridiumMessage(self).upgrade()
  File "iridium-parser.py", line 618, in upgrade
    return IridiumECCMessage(self).upgrade()
  File "iridium-parser.py", line 958, in upgrade
    return IridiumDAMessage(self).upgrade()
  File "iridium-parser.py", line 1011, in __init__
    the_crc=ida_crc16([chr(int(x,2)) for x in slice(crcstream,8)])
  File "/usr/local/lib/python2.7/dist-packages/crcmod/crcmod.py", line 447, in crcfun
    return fun(data, crc, table)
TypeError: argument 1 must be string or read-only buffer, not list

It doesn't happen with all lines in the output.bits file. Here's an example of what causes the error : 

RAW: i-1598877655-t1 0001166 1623596160 A:OK I:00000000026  89% 94.992 179 0011000000110000111100110011000101000101011001011111011101001001000010001101011111000110010101000010010100001000001000000100100101100010001100100001011000000000000000000000000000001001000100100000000000000000000000000000000000000000000000000110000000000000000000000000000000000000000000000000000110000000000000000000000010000100000011000000010000001000100011000100110011001000110000

Python 2.7.12
tried crcmod 1.7-2build2 from the Mint repo and also a freshly compiled 1.7 from pypi.org.

I've tried an output.bits from a previous install that I know parsed before and that fails in the same way. That makes me pretty sure that it's not the newly installed iridium-extractor that I have also just made.

:-(

Cheers,

Jem.

Sec42, 2020-09-01:
I just re-tested it with a git master (440c07f) - it still runs without error for me.

Cross-checking version numbers, I have crcmod 1.7 (just installed crcmod via pip2) and Python 2.7.18rc1, which doesn't seem that different.

I committed a patch that hopefully fixes it for you. Can you please test & report back?

Geezerjem, 2020-09-01:
No errors on previous or new output.bits file. Deffo fixed ty Sec42 and vm ! Which is excellent, but I have to ask why it just turned up for apparently no reason ?? Onwards and upwards !

Geezerjem, 2020-09-01:
vm as in very much, not virtual machine. Just wanted to clear that up.

Sec42, 2020-09-02:
that confirms that it was my crcmod change that broke it. Still confused why it worked for me but not for you. But glad it's fixed now. Sorry for the breakage.

closed 2020-09-02 by Sec42