muccc/iridium-toolkit #34: dr. Help

AdmiralGeta, 2020-10-08:
Hi muccc, 
I look at the bits using your toolkit and I can't understand how the bits after the parser convert the bits to HEX values ​​to get VOC data, tell me which way to look, I open the raw data and processed by your parser, but I don't see how they were converted, help me with this.

RAW
[image]

VOC
[image]

Sec42, 2020-10-09:
I'm not sure I understand your question. Are you asking on how the conversion from RAW: to VOC: (as pictured above) works?

class  `IridiumMessage`'s [`__init__`](https://github.com/muccc/iridium-toolkit/blob/6511c56b2827a92355f45d3f1cbb2704d4d8ad2a/iridium-parser.py#L322) method tries different decoding methods [until it figures out](https://github.com/muccc/iridium-toolkit/blob/6511c56b2827a92355f45d3f1cbb2704d4d8ad2a/iridium-parser.py#L362) that the packet has a valid Link Control Word (LCW). The [frametype field](https://github.com/muccc/iridium-toolkit/blob/6511c56b2827a92355f45d3f1cbb2704d4d8ad2a/iridium-parser.py#L569) of the decoded LCW (0) designates this as voice packet.
So `IridiumMessage`'s [`upgrade()`](https://github.com/muccc/iridium-toolkit/blob/6511c56b2827a92355f45d3f1cbb2704d4d8ad2a/iridium-parser.py#L620) method passes this packet on to the `IridiumVOMessage` class.

`IridiumVOMessage`'s [`__init__`](https://github.com/muccc/iridium-toolkit/blob/6511c56b2827a92355f45d3f1cbb2704d4d8ad2a/iridium-parser.py#L801) method checks whether this is a valid VDA, VO6 or VOD packet and [if not](https://github.com/muccc/iridium-toolkit/blob/6511c56b2827a92355f45d3f1cbb2704d4d8ad2a/iridium-parser.py#L823), just prints it's contents via the `IridiumVOMessage`'s [`pretty()`](https://github.com/muccc/iridium-toolkit/blob/6511c56b2827a92355f45d3f1cbb2704d4d8ad2a/iridium-parser.py#L837) method.

Hope that helps.

AdmiralGeta, 2020-10-12:
This helped me figure out how LCW works.

But I still don't understand how to descramble RAW data to VOC?

I would be very grateful to answer my mail
admiralgeta@gmail.com
best regard

Sec42, 2020-10-12:
I answered your question above to the best of my ability. Unless you have a more concrete question, I'm going to close this. Sorry.

closed 2020-10-12 by Sec42