muccc/iridium-toolkit #35: Pager Message Encrypted?

harrycheng20, 2021-01-17:
Hi, I am using gr-iridium to detect the iridium signals and decoding the messages with iridium-toolkit. Regarding the pager message, most of them look like this 
`5AFsISa5ywcAkFZxjrkd3pZCqDbz4DG*zTGE*N5+aKCmtPDJNBEVm5*x00WM`
Does this mean they have been encrypted? If it is not, I am wondering do we have any methods to decode them into readable words? 

FYI, I do saw a few  (1%) of the pager message readable, and they look like this: `nm@smtp.iwcs.net Test message sent from SNOC-CRON -> 881631099646@msg.iridium.com on Sun Jan 17 12:15:02 GMT 2021`

Thanks in advance.

schneider42, 2021-02-17:
To our knowledge the Iridium system itself does not provide encryption for pager messages. There are at least two different theories regarding these messages:

 1: They are from a third party, encoding/encrypting their messages before sending them to a pager. They'd need to be decoded/decrypted outside the pagers. Which third party might do such things are left to your imagination.
 2: These are actually not pager messages but for example related to Iridium Burst. I could imagine that they are re-using pager functionality and channels for this service.

harrycheng20, 2021-02-17:
Thank you for your reply! They are helpful.

closed 2021-02-17 by harrycheng20