muccc/iridium-toolkit #36: reassembler.py No Attribute 'signal'

notpike, 2021-05-02:
Hi,

Running into an issue with reassembler.py where I'm getting an... 
`AttributeError: 'MyObject' object has no attribute 'signal'`
on line 114. I didn't see any other references for signal in the MyObject class or in reassembler.py so I'm not quite sure where this value is being pulled from.  

Thanks!

skipak, 2021-05-02:
Which mode is the error on?  Others besides sbd?

notpike, 2021-05-02:
> Which mode is the error on? Others besides sbd?

msg and sbd.

Thank you.

Sec42, 2021-05-02:
Sorry, was a small bug on my end. Pushed a fix just now. Let me know if it works again for you now.

notpike, 2021-05-02:
No worries and thank you for maintaining this code! :D

notpike, 2021-05-02:
OK!
Tested it and it was successful. Thanks again for your help!

closed 2021-05-02 by notpike