muccc/iridium-toolkit #40: bch time

tianyasy, 2021-11-15:
the bch message brings the time( it seems like the frame_id), does it mean the time the sv build the bch frame?

Sec42, 2021-11-16:
Not sure what you mean by "bch". If you mean the time in the IBC frames: to our current knowledge that references the time of the start of the signal (preamble) of the start of the frame that contains the IBC packet.

And of course you then need to take the travel time of the signal from satellite to the receiver into account.

I have some changes to reassembler.py in the works that will hopefully clarify this a bit.

schneider42, 2022-02-01:
Addendum: It specifies the time at the start of the 90 ms L-Band frame that contained the IBC frame. The slot value tells you in which DL slot the IBC frame was sent.

The following code show cases the exact calculation. Please take into account that the time the signal took to reach you is not compensated.

https://github.com/muccc/iridium-toolkit/blob/39f12003fdee7494534d01398eecb45bc6b85d6e/reassembler.py#L520-L536

closed 2022-02-27 by Sec42