muccc/iridium-toolkit #45: Identifying parties during a call

Charlez700, 2022-01-31:
hello,

When exporting a pcap file (using the reassembler with the `lap` option), I've managed to identify one of the two parties in a phone call under the "Calling Party BCD Number" or the "Called Party BCD Number" entry. But I'm scratching my head on how can I identify the counterpart of the same call.

Not sure if it's something by design from the iridium network or I'm missing something elsewhere.

Kind regards,
Charlez.

Sec42, 2022-01-31:
I would expect the find "Calling Party BCD Number" in downlink frames representing incoming calls and "Called Party BCD Number" in uplink frames representing outgoing calls. So in both cases the number represents the "other" end of the call from the Iridium handset/phone.

As far as identification of the iridium handset/phone itself, you should be able to find a GSM-like imsi/timsi somewhere in that connection. I don't think the actual number is part of the protocol.

I have not done a lot of research in that area, so if you find some new/interesting things, let me know.

Charlez700, 2022-01-31:
Thank you! 
Regarding IMSI/TMSI data, the most similar data I found was the Iridium phone IMEI number. I'll record some test calls tomorrow and keep you posted.

Sec42, 2022-02-02:
I can see some IMSIs in the pcap:

`tshark -r output.pcap -Y "e212.imsi" -T fields -e frame.number -e gsm_a.dtap -e e212.imsi -E header=y`

Most seem to be part of `Location Updating Request`s, but I can also see some `Identity Response`s.

Charlez700, 2022-02-03:
that's pretty much correct. I did some more tests,  and when doing a call from the Iridium phone I've seen a CM Service Request packet (uplink) with the TMSI mobile identity number.

Still, I don't know how can I match both of the sides of the communication. I was expecting some sort of 'Assignment Command' packet [as on GSM](https://www.sharetechnote.com/html/BasicCall_Voice_GSM.html) to properly identify what's the ARFCN and TimeSlot of each party.

Sec42, 2022-02-05:
I don't have a complete picture on how the call setup works. But information about channel handover is part of the LCW, and (as far as I know) not inside the L3 GSM messages (i.e. the pcap). 

There is some optional content at the end of IBC frames called "channel assignment" which may be related/relevant here.

If you want to go deeper into this, I suggest we chat via some other medium. Shoot me an email, and we can connect on something more realtime like IRC or Threema if you wish.

closed 2022-02-27 by Sec42