muccc/iridium-toolkit #50: Error while reassembling using sbd mode

amitpathania, 2022-02-25:
while running 
reassembler.py -i output.parsed -m sbd
getting this error:
`
Traceback (most recent call last):
  File "reassembler.py", line 1853, in <module>
    zx.run(fileinput.input(ifile))
  File "reassembler.py", line 201, in run
    self.consume(mo)
  File "reassembler.py", line 1090, in consume
    self.consume_l2(zz)
  File "reassembler.py", line 1253, in consume_l2
    ult,prehdr.hex(":"),ascii(data, escape=True)))
TypeError: hex() takes no arguments (1 given)
`

Sec42, 2022-02-25:
Bytes.hex(argument) is a feature that is available since python 3.8. I suspect your python version is too old.

I might implement a workaround for older python versions at some point. But if you want it to work right now, I suggest upgrading python.

Sec42, 2022-02-27:
I've just pushed a workaround with 036fa947, let me know if it works for you.

And for your question - as far as i can tell: no