muccc/iridium-toolkit #51: live parsing

amitpathania, 2022-03-01:
If there any option to carryout online/live parsing of output bits (using iridium toolkit) produced by iridium-extractor to get voice/sms in real time. Presently, first you have to write output bits into file and then parse that file using iridium-parser and then extract voice. Anyone who has worked on solution to provide realtime parsing?

Sec42, 2022-03-01:
Live parsing is not really the main focus of iridium-toolkit. The whole code is a work in progress & built in a way to be able to run again over old data after changes to the parser or reassembler are made.

That said, SMS are part of the LAPDm layer2 messages (IDA). There is `reassembler -m gsmtap` which will help you feed them live into a running wireshark instance. Getting things like SMS out of wireshark/tshark live should be possible, but is left as an exercise for the reader :-)

Voice doesn't really make sense live IMO. Playing it live over speaker won't work as soon as there's more than one call going on. And if you're just creating files, you can run that whenever you're ready to look at it. All in all, there is no code to facilitate that. You would also need to somewhat reliably detect when a call is ending.

Let us know if you write any code do any of that, though.

closed 2022-03-01 by Sec42

amitpathania, 2022-03-02:
one can parse the pcap file created by reassembler -m gsmtap using following filters in tshark to get sms text:
`
tshark -r output.pcap -Y gsm_a.dtap -T fields -e gsm_a.dtap gsm_sms.tp-oa -e gsm_sms.scts.year -e gsm_sms.scts.month -e gsm_sms.scts.day -e gsm_sms.scts.hour -e gsm_sms.scts.minutes -e gsm_sms.scts.seconds -e gsm_sms.sms_text
`