muccc/iridium-toolkit #53: Creating kml file

amitpathania, 2022-03-02:
I am trying to convert IRA frames to a kml file to be viewed in google earth using following commands:
Run as grep ^IRA output.parsed |perl mkkml tracks > output.kml to display satellite tracks
Run as grep ^IRA output.parsed |perl mkkml heatmap > output.kml to create a heatmap of sat positions and downlink positions

But I'm getting errors and kml file is not getting created. I am getting following error when using heatmap command:
`
Couldn't parse: IRA: p-1642567800 010218310.0000 1626250624  96%   0.002 098 DL sat:071 beam:39 xyz=(+0293,+1239,+0957) pos=(+36.93/+076.70) alt=015 RAI:48 ?00 bc_sb:07
Couldn't parse: IRA: p-1642567800 010221366.0000 1626317440  97%   0.002 119 DL sat:067 beam:27 xyz=(+0475,+1632,+0558) pos=(+18.17/+073.77) alt=800 RAI:48 ?10 bc_sb:29 FILL=1
Couldn't parse: IRA: p-1642567800 010225686.0000 1626317312  98%   0.002 096 DL sat:067 beam:27 xyz=(+0309,+1328,+0824) pos=(+31.15/+076.90) alt=017 RAI:48 ?10 bc_sb:29
Couldn't parse: IRA: p-1642567800 010229735.0000 1626317056  94%   0.002 095 DL sat:067 beam:28 xyz=(+0474,+1627,+0573) pos=(+18.68/+073.76) alt=800 RAI:48 ?00 bc_sb:19
Couldn't parse: IRA: p-1642567800 010230006.0000 1626316928 100%   0.003 093 DL sat:067 beam:27 xyz=(+0474,+1627,+0574) pos=(+18.71/+073.76) alt=801 RAI:48 ?10 bc_sb:29
Down maxval: 0.000
Use of uninitialized value $sc in numeric gt (>) at mkkml line 225.
Up maxval: 0.000
Use of uninitialized value $sc in numeric gt (>) at mkkml line 225.`
 I am getting following error when using mkkml tracks command:
`
Couldn't parse: IRA: p-1642567800 010216775.0000 1626317824  87%   0.001 096 DL sat:067 beam:28 xyz=(+0311,+1426,+0640) pos=(+23.68/+077.70) alt=019 RAI:48 ?00 bc_sb:19 INCOMPLETE
Couldn't parse: IRA: p-1642567800 010217046.0000 1626317824  98%   0.003 095 DL sat:067 beam:27 xyz=(+0311,+1335,+0812) pos=(+30.64/+076.89) alt=017 RAI:48 ?10 bc_sb:29
Couldn't parse: IRA: p-1642567800 010218310.0000 1626250624  96%   0.002 098 DL sat:071 beam:39 xyz=(+0293,+1239,+0957) pos=(+36.93/+076.70) alt=015 RAI:48 ?00 bc_sb:07
Couldn't parse: IRA: p-1642567800 010221366.0000 1626317440  97%   0.002 119 DL sat:067 beam:27 xyz=(+0475,+1632,+0558) pos=(+18.17/+073.77) alt=800 RAI:48 ?10 bc_sb:29 FILL=1
Couldn't parse: IRA: p-1642567800 010225686.0000 1626317312  98%   0.002 096 DL sat:067 beam:27 xyz=(+0309,+1328,+0824) pos=(+31.15/+076.90) alt=017 RAI:48 ?10 bc_sb:29
Couldn't parse: IRA: p-1642567800 010229735.0000 1626317056  94%   0.002 095 DL sat:067 beam:28 xyz=(+0474,+1627,+0573) pos=(+18.68/+073.76) alt=800 RAI:48 ?00 bc_sb:19
Couldn't parse: IRA: p-1642567800 010230006.0000 1626316928 100%   0.003 093 DL sat:067 beam:27 xyz=(+0474,+1627,+0574) pos=(+18.71/+073.76) alt=801 RAI:48 ?10 bc_sb:29`

Sec42, 2022-03-02:
This script is quite old and not really maintained. I have quickly updated the IRA regex so it runs now. Haven't really had the time to test the functionality.

amitpathania, 2022-03-03:
Thanks for quick update. Now both commands are running without error but output.kml for `perl mkkml tracks` is having errors. However, output.kml for `perl mkkml heatmap` is working fine.
Thanks

Sec42, 2022-03-16:
I just tested all three formats (tracks/heatmap/beams) with local google-earth (v 7.3.4) and could view them without problems. What tool are you using?

amitpathania, 2022-03-29:
Sorry for late reply. Its working. Closing the issue

closed 2022-03-29 by amitpathania