muccc/iridium-toolkit #58: Information about iridium frames

michael040101, 2022-03-10:
Hi!
I’m trying to find some information about the iridium frames and more specifically on all the tones and unique words.
I’ve seen some in the wiki.muc.ccc but it’s very partial.
Can you help me with that?

Sec42, 2022-03-12:
Not really sure what kind of information you are looking for, but your best bet are the talks we gave:

newest, most complete is from HOPE:
https://www.youtube.com/watch?v=JhJT7Cvh6NE 

older, so less info, but more focus on the signal parts is from CCC Congress 2014:
https://media.ccc.de/v/31c3_-_6236_-_en_-_saal_1_-_201412281645_-_iridium_pager_hacking_-_sec_-_schneider

If you have more specific questions, let me know.

michael040101, 2022-03-12:
Do you maybe have a list of all the different tones and unique words that the system uses?

michael040101, 2022-03-15:
Can you share some documents that summarizes the information please?

schneider42, 2022-03-15:
@michael040101 I'm sorry we can't. The code and the presentations above are the documentation. Maybe this will change at some point.

Regarding your questions before: I'm not aware of "tones" being used in Iridium. The only two unique words relate to the UW for uplink and downlink frames and can be found here: https://github.com/muccc/gr-iridium/blob/master/lib/iridium.h#L24

michael040101, 2022-03-16:
I saw online that the duration of simplex signal, containing preamble + UW + data, is 20.32ms.
Can you confirm that?
Because if I look at the sizes of the simplex you provided the numbers don’t add up to that value.

schneider42, 2022-03-16:
(64 + 444) symbols / (25000 symbols/second) = 20.32 ms

michael040101, 2022-03-17:
The UW length which is 12 symbols,  is considered to be inside of the preamble length which is 64?

schneider42, 2022-03-17:
the preamble is an unmodulated carrier and does not include the UW. See this slide: https://media.ccc.de/v/31c3_-_6236_-_en_-_saal_1_-_201412281645_-_iridium_pager_hacking_-_sec_-_schneider#t=507

michael040101, 2022-03-17:
My intention is that inside the iridium.h file you defined 2 different parameter - preamble length and UW length.
And when you answered one of my questions about the simplex time duration you added the preamble length to the data length, but didn’t consider the UW length.
Maybe the 12 symbols of the UW is considered inside the 444 symbols of the data?

schneider42, 2022-03-17:
Yes, in the calculation above and inside the gr-iridium code, the UW is considered to be part of the "frame"

amitpathania, 2022-04-26:
What are IBC, RAW, ISY, STL, IMS and VOD Iridium frame types contained in parsed file?Any documentation available

schneider42, 2022-04-26:
https://youtu.be/cvKaC4pNvck?t=618

That is the most up to date thing we have that is not code :)

closed 2023-06-04 by Sec42