muccc/iridium-toolkit #80: Position of IRA

tk240, 2022-03-23:
Output of IRA signals pos=(+49.51/+029.99) (I assume it is in decimal degrees, please tell me if not). 
The position information is printed with only two decimals after '.'  . Is it rounded in the software?  I wonder how many decmals possible to reach? Is it possible to get 5 decimal aftter '.' ?

Sec42, 2022-03-23:
The position transmitted by the satellites is in the xyz= info. This is in the ECEF coordinate system with 4km granularity in all three axis. So no, there is not enough precision available for 5 decimals.

The lat/lon value in this line is only a quick approximation for convenience, see [bitsparser.py#L1385](https://github.com/muccc/iridium-toolkit/blob/master/bitsparser.py#L1385).

For a more exact conversion check [Geographic_coordinate_conversion](https://en.wikipedia.org/wiki/Geographic_coordinate_conversion#From_ECEF_to_geodetic_coordinates), for example via the pyproj library as used in [locator.py](https://github.com/muccc/iridium-toolkit/blob/master/locator.py#L12-L17).

closed 2022-03-23 by Sec42

tk240, 2022-03-25:
Does this mean that the position may have +-4km error for each axis?