muccc/iridium-toolkit #82: Information contained when running reassembler using page mode

amitpathania, 2022-04-10:
I am running reassembler.py in paging requests (Ring Alert Channel) mode using -m page :
`python3 reassembler.py -i  output.parsed -m page`
I am getting following output:
'109 22  18.01  59.27 798 : 109f0b0c 02
085 22  21.96  67.98 018 : 109f104c 02
085 22  24.46  68.19 020 : 0d011548 11
085 23  18.51  57.27 800 : 0d011548 11
085 23  26.66  70.00 018 : 836951fc 21
085 36  34.00  57.39 798 : 0ca66178 05
065 03  27.14  71.88 020 : 836951fc 21
087 23  20.31  54.93 800 : 109f104c 02
029 12  27.95  76.69 019 : 0ca66178 05
068 20  14.87  82.96 800 : 02a27760 05
048 26   7.97  78.38 803 : 0ca1eb7b 05
071 26  18.59  75.90 019 : 101ad054 02
071 26  27.35  75.81 018 : 0d011548 05
071 27  18.27  76.07 801 : 0cf48243 05
071 26  32.42  75.88 016 : 0cf48243 05`

I am not able to understand information contained in the output? Can someone explain what these values mean?

Sec42, 2022-04-11:
What have you tried to understand that info?

'page' mode is just printing the page requests from IRA packets in a slightly different form - see https://github.com/muccc/iridium-toolkit/blob/master/reassembler.py#L1573
it is one line per page with satid, beamid, latitude, longitude, altitude before the colon, and the page request afterwards. The page request identifies the mobile device to wake up & respond.

'msg' is (reassembling) and printing the pager messages you see in MSG packets. What you see there is what is sent over the air as is. There are very few plaintext page messages remaining as iridium is not selling/marketing that service for several years now.

The encoded text is probably part of the "Iridium Burst" service. But it's contents are not yet understood.

closed 2022-04-11 by Sec42

amitpathania, 2022-04-15:
Due to lack of proper documentation, many times it becomes difficult to understand the info contained in the output especially when its encoded or doesn't make any logical sense. You yourself have written that contents of encoded text are not yet understood. Therefore, if someone asks a simple question, there is no need to be rude. Your project has helped many people like me to understand dynamics of Iridium satellite communication and we all really appreciate your efforts to keep project updated and your prompt replies. Thanks

Sec42, 2022-04-15:
I'm sorry you feel that my answer was rude. I did answer both your questions.

If you feel a reference to the source is rude, please consider that this is not an end-user ready tool -- and in my experience source code is an easy way to communicate for developers