muccc/iridium-toolkit #83: Reassembler.py Module Not Found Error

muaddib1984, 2022-05-14:
I just cloned the latest version of iridium-toolkit and reassembler.py errors
```
Traceback (most recent call last):
  File "/home/user/iridium-toolkit/reassembler.py", line 12, in <module>
    import iridium.config
ModuleNotFoundError: No module named 'iridium.config'
```
I haven't had this issue before, but i don't think i've used the newest version since the iridium.config option was added. 
I'm using python 3.9.5

Sec42, 2022-05-14:
I can not reproduce this error.

Can you try with a fresh clone?

muaddib1984, 2022-05-15:
i just tried with a fresh clone. same issue. it seems that the implicit namespace for iridium/config is not being picked up. I get that this is more of a python issue, but its a new install of ubuntu 21, i have gr-iridium and GNURadio installed. the extractor and parser work fine.

muaddib1984, 2022-05-15:
ok, i got it to work by uninstalling gr-iridium. it appears theres a namespace conflict when installing gr-iridium system-wide. I think you guys recommend (and i assume use) PyBombs, so maybe that's the issue for me. I'm installing GNURadio and gr-iridium systemwide, so when python tries to import iridium.config, it finds  gr-iridium's namespace instead of the iridium-toolkit module ```iridium-toolkit/iridium/config.py```
for a proposed fix, maybe changing the module under iridium-toolkit to something other than iridium might work?

Sec42, 2022-05-15:
Thanks for finding the cause. That's unfortunate. I would have thought that the local directory had precedence over system files. Ah well, I'll probably have to rename the thing.

schneider42, 2022-05-16:
With Ubuntu 22.04 we will have more situations where gr-iridium is installed system wide.

alphafox02, 2022-05-21:
I had to revert to f1be9e5 on DragonOS Focal (20.04) to make it work again, of course it’s using GR 3.8 branch on the desktop build. I had just uploaded a new image and did a last minute pull on the toolkit (forgot to check it).

closed 2022-05-21 by Sec42

muaddib1984, 2022-05-21:
@alphafox02 if you follow my instructions above it will work fine. Just change the directory and do a find/replace on the references.

Sec42, 2022-05-21:
@muaddib1984 / @alphafox02 this issue should be fixed by the above commit. Let me know if it doesn't work for you.

alphafox02, 2022-05-21:
Switched to master and did a pull, works perfectly fine now. Thank you! I’ll change that to the default again in the next iso.

muaddib1984, 2022-05-21:
Cool. I'll try it later today. Thanks @schneider42 and @Sec42 !

commit a957f287 (2022-05-21): [reassembler] fix #83: rename iridium -> iridiumtk