muccc/iridium-toolkit #87: Is there some way to get the position of the ISU(iridium terminal)?

tianyasy, 2022-07-19:
In SBD, 0x070500 can show the position information, but it seems not like the position of the ISU

tianyasy, 2022-07-19:
and in LCW, the contents of geo-loc is all 0s.

tianyasy, 2022-07-20:
there is a alogrithm tdoa chan. but I don't kown the meaning of dtoa from lcw type maint[2]

tianyasy, 2022-07-20:
thre is some datas from lcw maint[2]: 
lcw, dir_1, f_228, ch_178, sn_2804, maint[2], dtoa_2, dfoa_0.
lcw, dir_1, f_453, ch_178, sn_4985, maint[2], dtoa_1, dfoa_0.
lcw, dir_1, f_586, ch_178, sn_6768, maint[2], dtoa_0, dfoa_5.
lcw, dir_1, f_590, ch_178, sn_6867, maint[2], dtoa_0, dfoa_4.
lcw, dir_1, f_594, ch_178, sn_7024, maint[2], dtoa_0, dfoa_4.
lcw, dir_1, f_598, ch_178, sn_7101, maint[2], dtoa_0, dfoa_2.
lcw, dir_1, f_602, ch_178, sn_7150, maint[2], dtoa_127, dfoa_3.
lcw, dir_1, f_606, ch_178, sn_7191, maint[2], dtoa_0, dfoa_2.
lcw, dir_1, f_610, ch_178, sn_7299, maint[2], dtoa_1, dfoa_7.
lcw, dir_1, f_614, ch_178, sn_7359, maint[2], dtoa_0, dfoa_6.
lcw, dir_1, f_622, ch_178, sn_7505, maint[2], dtoa_0, dfoa_6.
lcw, dir_1, f_626, ch_178, sn_7575, maint[2], dtoa_127, dfoa_1.
lcw, dir_1, f_630, ch_178, sn_7637, maint[2], dtoa_1, dfoa_3.
lcw, dir_1, f_634, ch_178, sn_7798, maint[2], dtoa_0, dfoa_2.
lcw, dir_1, f_638, ch_178, sn_7942, maint[2], dtoa_127, dfoa_2.
lcw, dir_1, f_642, ch_178, sn_8068, maint[2], dtoa_1, dfoa_2.
lcw, dir_1, f_646, ch_178, sn_8143, maint[2], dtoa_127, dfoa_0.
lcw, dir_1, f_650, ch_178, sn_8247, maint[2], dtoa_1, dfoa_0.
lcw, dir_1, f_654, ch_178, sn_8301, maint[2], dtoa_127, dfoa_0.
lcw, dir_1, f_658, ch_178, sn_8358, maint[2], dtoa_1, dfoa_1.
lcw, dir_1, f_662, ch_178, sn_8409, maint[2], dtoa_0, dfoa_0.
lcw, dir_1, f_670, ch_178, sn_8520, maint[2], dtoa_0, dfoa_1.
lcw, dir_1, f_674, ch_178, sn_8637, maint[2], dtoa_0, dfoa_0.
lcw, dir_1, f_678, ch_178, sn_8745, maint[2], dtoa_0, dfoa_127.
lcw, dir_1, f_686, ch_178, sn_9035, maint[2], dtoa_127, dfoa_126.

tianyasy, 2022-07-20:
so, in the data showed above, what mean 2 in dtoa or 1....127?

Sec42, 2022-07-28:
Github issues are not intended for questions.
Additionally you put two different questions in this thread.

To answer the original question: I am not aware of any Iridium phone or SBD receiver sending their position.
This would require them to have a GPS which (to my knowledge) they do not have.

All positions that the toolkit decodes are sent from the satellite and refer to either the satellite position, or the estimated center point of a spot-beam (which is approx 400km in diameter).

closed 2022-07-28 by Sec42