muccc/iridium-toolkit #98: Issue with Parsing output.bits

CurtisDufour, 2023-02-26:
I have been successful with pulling Iridium data with gr-iridium, and have data in the output.bits file. I am running the following command:

```
pypy <my directory>/iridium-toolkit/iridium-parser.py <my directory>/gr-iridium/output.bits > <my directory>/iridium-toolkit/output.parsed 
```
This, according to what I am reading on the readme, should pull the data from output and parse it. Instead, I am receiving the following error:

```
File "<my directory>/iridium-toolkit/iridium-parser.py", line 119
    print("WARN: no sigmf-meta source file. Using (probably-wrong) hardcoded defaults", file=sys.stderr)
                                                                                            ^
SyntaxError: invalid syntax (expected ')')
```

I don't know enough here to know where to find the hardcoded defaults, or which ones it's talking about. I also can't find what's spurring that SyntaxError. Apologies if this is user error.

schneider42, 2023-02-26:
Which Python version do you have on your system?

CurtisDufour, 2023-02-27:
Python is:
Python 3.11.2

I'm also noticing that pypy has the following version information:
Python 2.7.18 (7.3.3+dfsg-2, May 02 2021, 15:09:28)
[PyPy 7.3.3 with GCC 10.2.1 20210110]

I do have pypy3 on my system as well:
Python 3.9.16 (7.3.11+dfsg-2, Feb 06 2023, 17:14:22)
[PyPy 7.3.11 with GCC 12.2.0]

And, surprise! all I had to do was install python3-crcmod, and it appears that I have data in output.parsed, and I'm no longer getting error codes. Thank you for your assistance.