muccc/iridium-toolkit #99: ambe codec not found

rustjakey, 2023-03-02:
Same issue as this closed post [](https://github.com/muccc/iridium-toolkit/issues/42)

Have tried the suggested solution but no luck.

Any help greatly appreciated. 

[image]

[image]

schneider42, 2023-03-06:
try "export PATH=`...

rustjakey, 2023-03-10:
After doing that, the terminal session cant start py. 

If I start a new terminal, the PATH changes are reset but I am back at square one. 

Thanks

[image]

rustjakey, 2023-03-10:
Seem to have fixed that issue with `export PATH="/usr/bin:$PATH"`

rustjakey, 2023-03-10:
Worked once. This is the path now
[image]

rustjakey, 2023-03-10:
I cleared all the changes I made to the env variable files and put it in the bashrc one. Works. Now another Issue I will open separately.

closed 2023-03-10 by rustjakey