muccc/iridium-toolkit #100: Error ratio: nan

rustjakey, 2023-03-10:
When selecting any of the data points left and right this happens. Long or short.

[image]

Sec42, 2023-04-06:
I'm not 100% sure I remember correctly. But Error ratio is defined as frames_ok / frames_total
A result of nan would happen if frames_total were 0, so no frames were sent/piped into ir77_ambe_decode.

I just tested it on master, and I couldn't find any obvious problems. Which version are you running?

If I were to guess, either you're not selecting any packets, or your bits format is not being accepted by parts of the tools (i.e. bits_to_dfs.py).

alexviterni, 2024-02-03:
hello dear
In ITL frames
In the code available in the iridium toolkit, the bitsparser section
ITL frame header as
header_time_location="11"+"0"*94

While in the movie
((Iridium Satellite Hacking HOPE XI 2016))
as
"11"+"0"*350
is. Now which one is correct?

Sec42, 2024-02-05:
Both :)

Check https://github.com/muccc/iridium-toolkit/blob/master/itl.py#L4 and https://github.com/muccc/iridium-toolkit/blob/master/bitsparser.py#L748 

The format has simply changed over time :-)

closed 2024-02-05 by Sec42