muccc/iridium-toolkit #103: parsing out ADS-B data

mebrown47, 2023-06-03:
Looking at iridium-parser.py and reassembler.py I am unclear where I might add additional message types. Would this be an additional frame class in iridium-parser.py or a refinement of SBD in reassembler.py? Attempting to decode ADS-B data carried on Iridium (Aireon).

Sec42, 2023-06-04:
It depends on what data you mean? The current ACARS parsing (i.e. the output of "reassembler.py -m acars") is implemented in [ReassembleIDASBDACARS](https://github.com/muccc/iridium-toolkit/blob/master/iridiumtk/reassembler/sbd.py#L231) - not sure if you're talking about some sub-type that's parsed incorrectly or some new yet unknown packet? Might be easiest if you could provide examples...

mebrown47, 2023-06-04:
I'm referring specifically to the Aireon "space-based ADS-B" system deployed on the Iridium NEXT sats. No luck finding detailed technical specs but it appears to be a SBD payload the format of which I'm unable to find documented. Based on your reply (thanks for that) it would be handled in reassembler.py and/or sbd.py

Not sure if that is enough of an example to clarify. Maybe a better way to ask the question is where in the current Iridium workflow would a new SBD payload type be identified/parsed/decoded?

Sec42, 2023-06-05:
To answer your question: except for ACARS there is no other SBD device-specific protocol that we decode. So a new one would probably start out in the sbd.py file, or possibly if it is complex/big enough would get moved out to a new file.

But let me address your assumptions a bit:

SBD is the protocol that is being talked between iridium-enabled devices and the satellite.

If you are referring to the space-based ADS-B receiving payload that was put on the iridium next satellites: To my knowledge this has nothing to do with iridium. These are merely (custom) receivers that collect the ADS-B information and send them down to the company that "paid" for them. My understanding is also that they are completely "passive" listening devices. So I don't expect to see any of that data in the "normal" iridium communication - I don't see any reason for that data  being sent/relayed to normal iridium users.

It is most likely getting sent via the inter-satellite links and then on the downlink to the iridium gateway in the US - that is something we have no access to and would be a completely different project.

However, I may of course be wrong. If you have captured data that contains such information, let me/us know, and we'll take a look at it.

closed 2023-06-21 by Sec42

Hieuwwa, 2024-05-17:
Hello. I'm also wondering if Iridium Next's parse file data includes ADS-B data. And if so, in which frame (IIQ, or IIP) is it located?