muccc/iridium-toolkit #105: gsm downlink 951.4 Mhz can not be parsed with hackrfone

arifkyi, 2023-09-05:
Hello Sir,

Need your kindly help, after i scanning the gsm with gr-gsm tools 

grgsm_scanner  run with hackrfone

ARFCN:   26, Freq:  940.2M, CID: 63012, LAC:  3613, MCC: 510, MNC:  21, Pwr: -54
ARFCN:   80, Freq:  951.0M, CID: 26771, LAC:  3613, MCC: 510, MNC:  21, Pwr: -51
**ARFCN:   82, Freq:  951.4M, CID: 32772, LAC:  3725, MCC: 510, MNC:  21, Pwr: -20.  --> i choose this**
RFCN:   84, Freq:  951.8M, CID: 65063, LAC:  3725, MCC: 510, MNC:  21, Pwr: -44
ARFCN:   86, Freq:  952.2M, CID:  7242, LAC:  3613, MCC: 510, MNC:  21, Pwr: -46
ARFCN:   91, Freq:  953.2M, CID: 64917, LAC: 24159, MCC: 510, MNC:  11, Pwr: -43

then i decide use the 951.4M because the strongest one,

then i make this config:
cat gr-iridium/examples/hackrf10msp_951.4.conf
[osmosdr-source]

device_args='hackrf=0'

# Use the following line to turn on the antenna bias
#device_args='hackrf=0,bias=1'

**sample_rate=10000000**
**center_freq=951400000**
bandwidth=9000000

# Turn on the pre-amp
gain=14

# Moderate gains
if_gain=40
bb_gain=20

then i run this command iridium-extractor -D 4 gr-iridium/examples/hackrf10msp_951.4.conf > output.bits
attached below is the output.bits
[output.bits.gz](https://github.com/muccc/iridium-toolkit/files/12521902/output.bits.gz)

and then i leave it around 15 to 20 minutes
and then run these commands below respectively:
python3 iridium-parser.py -p output.bits > output.parsed   --> found the output.parsed is zero
run again withouth -p
python3 iridium-parser.py output.bits > output.parsed  
and then run again the output.parsed with, found some result

but i noticed a lot of like this below:
RAW: p-1693905457-e000 000946498.6657 0951082315  67% -47.37|-087.90|21.39 0011011000110010 1011001100111100 0110100000110101 00001100 ERR:Message: Access code missing

both result of command below is failed,
rifky@rifky-virtual-machine:~/Desktop/iridium-toolkit$ ./reassembler.py -i output.parsed -m ida
Kept 0/25073 (0.0%) lines
0 valid packets assembled from 0 fragments (1:0.00).
0/0 (0.0%) broken fragments.
0 dupes removed.
rifky@rifky-virtual-machine:~/Desktop/iridium-toolkit$ ./reassembler.py -i output.parsed -m lap
Kept 0/25073 (0.0%) lines
0 valid packets assembled from 0 fragments (1:0.00).
0/0 (0.0%) broken fragments.
0 dupes removed.

My goal to see the "lap" or IDA" mode in pcap file for the gsm,  Need your kindly help for your suggestion and direction sir?
br,
rifky

arifkyi, 2023-09-05:
i also attach the file for the output.parsed.gz  just in case need it for the diagnose
[output.parsed.gz](https://github.com/muccc/iridium-toolkit/files/12522076/output.parsed.gz)

arifkyi, 2023-09-05:
Hello,

thanks for your swift response Sir,

yes my confusion started from this guidance:

Supported modes are currently:

- ida - outputs Um Layer 3 messages as hex

- lap - GSM-compatible L3 messages as GSMtap compatible .pcap

thank you very much for the clarity,
i think we can close this issue, thanks

Sec42, 2023-09-06:
Thanks for the pointer. I will maybe re-word this to make it a bit clearer.

closed 2023-09-06 by Sec42

arifkyi, 2023-09-13:
> Thanks for the pointer. I will maybe re-word this to make it a bit clearer.

the pleasure is mine, thanks for clarifying 🙏🙏