muccc/iridium-toolkit #114: Information about IRA tracks

cyqsir, 2024-04-12:
https://github.com/muccc/iridium-toolkit/blob/c1cb31b517752b145c0f2b9dc077d539230ebcb9/bitsparser.py#L1509    The value of ra_pos_y in the formula seems to be no greater than 4096→int(self.bitstream_bch[26:37],2) - int(self.bitstream_bch[25])*(1<<11)   ；In tle, some Iridium orbits can reach about 6000km, whether there is some conflict, or the formula is not the position under ecef.

cyqsir, 2024-04-12:
The value of ra_pos_y in the formula seems to be no greater than 4096：Let me change that here： The value of ra_pos_y ranges from -2048 to 2048，In tle, some Iridium orbits can reach about 6000km

schneider42, 2024-04-12:
The position is encoded in 4 km steps. You have to multiply these numbers with 4000 to get to the XYZ (ECEF) coordinate in meters.

cyqsir, 2024-04-12:
Does that mean the orbit could be off by 4km?

> The position is encoded in 4 km steps. You have to multiply these numbers with 4000 to get to the XYZ (ECEF) coordinate in meters.

Does that mean the orbit could be off by 4km

Sec42, 2024-04-12:
Yes, the precision of this field is 4km in x/y/z, so the position can be off by as much as sqrt(3)*4000m.

You can see that the lat/lon/alt conversion (e.g. at https://github.com/muccc/iridium-toolkit/blob/master/bitsparser.py#L1521 ) contain a "*4" for the altitude (in km).

cyqsir, 2024-04-13:
> Yes, the precision of this field is 4km in x/y/z, so the position can be off by as much as sqrt(3)*4000m.
> 
> You can see that the lat/lon/alt conversion (e.g. at https://github.com/muccc/iridium-toolkit/blob/master/bitsparser.py#L1521 ) contain a "*4" for the altitude (in km).

If I have a string of IRA signal demodulated bits, do not do other operations, such as adding time-stamped bits to bits, etc., and know the position of BPSK (iridium access) in bits, Use ‘’bitstream_bch=de_interleave3(bits(length(iridium access: length(iridium access)+96)) deinterleasing and self.ra_sat= int(self.bitstream_bch[0:7],2) # sv_id ‘’ does not get the correct value, how can I change it？What information is original IRA information?Does the deinterleaver length need to change if no additional information is added?

cyqsir, 2024-04-15:
> Yes, the precision of this field is 4km in x/y/z, so the position can be off by as much as sqrt(3)*4000m.
> 
> You can see that the lat/lon/alt conversion (e.g. at https://github.com/muccc/iridium-toolkit/blob/master/bitsparser.py#L1521 ) contain a "*4" for the altitude (in km).

https://github.com/muccc/iridium-toolkit/blob/c1cb31b517752b145c0f2b9dc077d539230ebcb9/bitsparser.py#L1506
Does the satellite id also need to be added a constant to be the real id, compared to TLE? And what's the constant? After all, [0:7] has a maximum of 127。

Sec42, 2024-04-15:
Have you watched one of the talks we gave at HOPE or CCC congress? That may answer some of the questions you have.

For the `sv_id` - the id there is (as far as we know) an internal number, the maximum is indeed 127 and seems to have no correlation to the official (TLE) names.
There is even a reassembler mode `satmap` that creates such a mapping, see https://github.com/muccc/iridium-toolkit/blob/master/iridiumtk/reassembler/id_sat_map.py .

Mahatma011, 2024-05-10:
Can you help me determine the coordinates of an Iridium subscriber?
Is this possible using the iridium toolkit?

bgp-sz, 2024-05-15:
This issue is about IRA packets - those contain positional information about the satellite position and the spot beam locations.

Sec42, 2024-05-15:
This issue is about IRA packets. Those contain positional info about the satellite and the spot beam.

Mahatma011, 2024-05-15:
Thank you very much for the quick response and help!
As I understand it, determining the location of a subscriber is possible and is determined using the position of the satellite and a spot beam?
Do I need to write separate code for this?
Or is this possible with the Iridium toolkit and should I look into the code for the IRA packages?
Thanks a lot!