muccc/iridium-toolkit #118: Receiving Parsing Error

RicardoCastanon, 2024-06-11:
I am running iridium-parser.py on some .bits files and each time the .parsed file shows this first row:
ERR:  ERR:Message: Couldn't parse: fft_burst_tagger :info: set_min_output_buffer on block 5 to 65536

Additionally, I am getting several RAW: rows, saying:
ERR:Message: Access code missing

Any help will be appreciated, thank you.

Sec42, 2024-06-12:
Both are normal.

"fft_burst_tagger :info: set_min_output_buffer on block 5 to 65536" is a message printed by gr-iridium. As it contains no bits, it's normal that iridium-parser cannot parse it.

The "Access code missing" messages indicate bits that do not contain the (error-free) access code, so it is telling you that it didn't parse these.
There is an option for iridium-parser `--uw-ec` which uses more cpu and tries to parse these lines that you can use. But be aware that it is more likely to contain further errors and may still fail parsing.

Alternatively you can also use `--errorfree` to just ignore all lines that can not be parsed.

closed 2024-06-12 by Sec42

Rango-T10000, 2024-10-28:
> I am running iridium-parser.py on some .bits files and each time the .parsed file shows this first row: ERR: ERR:Message: Couldn't parse: fft_burst_tagger :info: set_min_output_buffer on block 5 to [...]

Hello，have you solved this question? I met the same problem with you.

Rango-T10000, 2024-10-28:
> Both are normal.
>
> "fft_burst_tagger :info: set_min_output_buffer on block 5 to 65536" is a message printed by gr-iridium. As it contains no bits, it's normal that iridium-parser cannot parse it. [...]

Hi, does it mean that the burst signal received is too little to be parsed? Or the signal is too weak?