muccc/iridium-toolkit #119: Questions about the beam handover of Iridium satellites

xyj8, 2024-06-14:
Hello, I saw in your toolbox that you can calculate the transmitting beam of an Iridium signal. Have you ever compiled specific rules for the 48 beam changes of Iridium.

Sec42, 2024-06-14:
Hi,

we do not calculate any beams. We do print the beam number transmitted by the satellite in IRA and IBC packets.

The rule is that a satellite cycles through all it's enabled spot beams in 4.8 seconds, so if you check the receive timestamps of IRA packets from a single satellite modulo 4800ms you should be able to identify the order in which the beams are transmitted.

xyj8, 2024-06-15:
Hello, thank you for your response. I would like to know why the period is 4.8 seconds. The literature I have reviewed states that the period is 4.32 seconds.

Sec42, 2024-06-15:
You are correct, it was a typo. 48 * 90 ms = 4320ms

closed 2024-06-15 by Sec42