muccc/iridium-toolkit #120: ERR:Message: Can't parse problem!

Rango-T10000, 2024-10-28:
I use a IAA.01.121111 Iridium Certified Antenna connected to a LNA and USRP X310 to receive the iridium Signal via gr-iridium.
[image]

When I run the gr-iridium, the terminal ouput shows like this:
[image]
[image]

And I can get the ouput.bits file, then I use iridium_parser.py to parse this file: "pypy3 iridium-parser.py ./output.bits > output.parsed", however, the parse is totally failed! The oupur.parsed like this:
[image]

According to[ ](https://github.com/muccc/iridium-toolkit/issues/118), I try "--uw-ec" and "--errorfree" command, but still can not work. Could you please tell me what is the problem in my whole precess? Any advice? Really thanks.

donbmouse, 2024-11-11:
Hello,
I have used a HackRF setup of this toolkit as a control for over a year. But in the past couple of months, I have been seeing the same problem as Rango, above. Nothing in my rx chain has changed and the sky/location is the same. Is it possible the Iridium protocol has changed slightly? Or are you seeing packet consistency in other areas? I am in northern USA. 
Thanks

schneider42, 2024-11-17:
Can you share some screenshots to see how that interference looked like or what you did to remove it?